openscopeproject/InteractiveHtmlBom #458: Disable the display of certain fields in ibom

dmkuzmin, 2024-07-19:
I am trying to add fields to ibom using the options "--extra-fields References,Case,Part Number,Value_SUCH,Quantity" and "-show-fields References,Case,PartNumber,Value_SCH,Quantity". As a result, I have the necessary fields displayed and additionally what was the default "Reference,Value,Footprint,Quantity". How can I disable the display of certain fields? Well, it would be good not to show duplicates. I can disable the display in the dialog box.

qu1ck, 2024-07-19:
Don't use `--extra-fields` with `--show-fields`. Use `--show-fields` and `--group-fields`

From the cli help:
```
  --extra-fields EXTRA_FIELDS
                        Passing --extra-fields "X,Y" is a shortcut for --show-fields    
                        and --group-fields with values "Value,Footprint,X,Y" (default:  
                        None)
  --show-fields SHOW_FIELDS
                        List of fields to show in the BOM. (default: Value,Footprint)   
  --group-fields GROUP_FIELDS
                        Fields that components will be grouped by. (default:
                        Value,Footprint)
```

dmkuzmin, 2024-07-19:
Do the Reference and Quantity fields have an unchangeable position? At the beginning and end of the table?

dmkuzmin, 2024-07-19:
I understood. Thanks

closed 2024-07-19 by dmkuzmin