openscopeproject/InteractiveHtmlBom #463: when i use generate_interactive_bom in cadence skill, it retun nil

Jinying-Tang, 2024-08-14:
1、use python 3.12.5, generate generate_interactive_bom.exe
2、result = shell("generate_interactive_bom  xxx.json")
      result is nil

when i use generate_interactive_bom.exe generated with python3.11.9,   the result is t

qu1ck, 2024-08-14:
I don't have cadence and can't test the script. If you have issues with the script part then report them on the script repo https://github.com/juulsA/exportJson

If the script part is working and generates json but there is an issue with ibom then try running ibom from command line directly and show me any errors if you see them.

Jinying-Tang, 2024-08-15:
the script part is working and generates json success,  and running ibom from command line , Python 3.12 and 3.11 both executed successfully, with no error messages displayed
But when running in skill, python3.11 returns "t", python3.12 returns "nil", Although python3.12 returns "nil", it actually generates an htm file.

qu1ck, 2024-08-15:
> running ibom from command line , Python 3.12 and 3.11 both executed successfully, with no error messages displayed

Then there is no issue with ibom

> But when running in skill

That is also part of script that I can't test or do anything about since I don't have cadence.

Also it sounds to me like everything is working, why do you care what the result of that `shell()` call is?

Jinying-Tang, 2024-08-15:
> Then there is no issue with ibom
>
>
> That is also part of script that I can't test or do anything about since I don't have cadence.
>
> Also it sounds to me like everything is working, why do you care [...]

Because I want to display the corresponding prompt through the return result of calling shell()

qu1ck, 2024-08-15:
Well, alegro is so hostile that you can't even access their docs without a registration that requires a valid license so I can offer only generic debug advice: divide and isolate the issue into small bits until you pinpoint where exactly is the difference.

I have no idea what `shell()` return value is supposed to be and why it differs depending on python version used. If it's a direct call to the OS shell to execute an external command then it would make sense for it to be the return code of the process but it returning `t` in some cases makes no sense so ¯\\\_(ツ)\_/¯

Jinying-Tang, 2024-08-15:
> Well, alegro is so hostile that you can't even access their docs without a registration that requires a valid license so I can offer only generic debug advice: divide and isolate the issue into [...]

 Thank you for your patient reply, I can only ignore this issue for now.

closed 2024-08-15 by qu1ck