openscopeproject/InteractiveHtmlBom #465: Missing some Pin 1 silkscreen triangles

totallymindblown, 2024-08-18:
_Tested versions:_

**KiCad 8.0.4
InteractiveHtmlBom 2.9.0**

Some of the pin 1 triangles from silkscreen layer are not rendered.
This means that a clear orientation using **ibom.html** is not possible.

Example packages where pin 1 triangle does not show up in **ibom.html**:

- Package_TO_SOT_SMD:SOT-23-6
- Package_QFP:TQFP-100_14x14mm_P0.5mm

[image]

[image]

qu1ck, 2024-08-18:
Can you check with ibom from git latest? I think this was fixed in 002d62760ae2c91a267d8c8d2a3b5eccf81439a2

totallymindblown, 2024-08-18:
I downloaded hole repo and installed the plugin manually and it is working now.
Problem solved. 🍓 
Thank you. 😃

closed 2024-08-18 by totallymindblown

qu1ck, 2024-08-18:
Great.
I really should cut a release soonish...