openscopeproject/InteractiveHtmlBom #466: JSON Schema and DATAFORMAT don't clearly define DNP components

InRe-Dan, 2024-09-05:
It doesn't look like I can define a DNP component using a JSON file and the `generate_interactive_bom.py` script. `DATAFORMAT.md` implies that there can be a `bom` field on the `pcbdata` struct, but if I add such a field, the script gives me errors (fair enough - the schema doesn't define anything to do with `bom` or DNP components _as far as I can tell_). The `config` struct in `DATAFORMAT.md` also doesn't seem to be something that I can define in the JSON either.

What am I missing here? Or are these features not supported via the script?

qu1ck, 2024-09-05:
Bom field in pcbdata is generated from your input and config, it is intentionally not in the json schema.

To have a DNP component you just have to add an extra field "dnp": "true" and then use that field with `--dnp-field=dnp` on command line.
https://github.com/openscopeproject/InteractiveHtmlBom/blob/2b8dfa665e2ae43fd47efbf38d0708cbab2a5413/InteractiveHtmlBom/ecad/schema/genericjsonpcbdata_v1.schema#L48

InRe-Dan, 2024-09-05:
Brilliant, thank you so much.

closed 2024-09-05 by InRe-Dan