openscopeproject/InteractiveHtmlBom #467: Crashes after installing pymupdf

denisbondar, 2024-09-16:
Hi,
The pymupdf extension was installed for the Board2Pdf plugin to work. After that when I try to launch Interactive Html Bom my KiCad just closes instantly.
I checked this by reinstalling KiCad. The problem definitely appears after installing pymupdf.

I have macOS and I installed pymupdf as follows:
```shell
cd /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions
bin/pip3 install -U pymupdf
```

Here's a list of all the extensions installed:
```shell
$ bin/pip3 list
Package            Version
------------------ -----------
attrdict           2.0.1
certifi            2024.8.30
charset-normalizer 3.3.2
idna               3.8
packaging          24.1
pip                24.2
pyclipper          1.3.0.post5
PyMuPDF            1.24.10
PyMuPDFb           1.24.10
requests           2.32.3
setuptools         58.1.0
sip                6.8.3
six                1.16.0
tomli              2.0.1
urllib3            2.2.2
wheel              0.44.0
wxPython           4.2.2a1
```

Can you tell me how I can find more data about the problem to provide to you?

qu1ck, 2024-09-17:
Does the problem go away if you uninstall pymupdf?
Do other plugins with GUI work? Does kicad's built in scripting console work?

How exactly did you determine that it's particularly that package that causes the crash? My plugin does not use any extra packages, I suspect something else is going on.

denisbondar, 2024-09-17:
> Does the problem go away if you uninstall pymupdf?

Yes. As soon as I remove PyMuPDF, your plugin works fine again.
```shell
cd /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.9
bin/pip3 uninstall PyMuPDF PyMuPDFb
```
After removing the PyMuPDF, your plugin starts working normally.

> Do other plugins with GUI work? Does kicad's built in scripting console work?

Yes. The other plugins display their windows and work fine.
Yes. The Python script console is displayed correctly.

> How exactly did you determine that it's particularly that package that causes the crash? My plugin does not use any extra packages, I suspect something else is going on.

Extension PyMuPDF is required for the **Board2Pdf** plugin to work. After installing PyMuPDF your plugin stops working. After uninstalling PyMuPDF your plugin workins again.

qu1ck, 2024-09-17:
Interesting, I can reproduce. Crash happens in pymupdf lib
<details>
<summary>Full crash log</summary>

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               kicad [1099]
Path:                  /Applications/KiCad/KiCad.app/Contents/MacOS/kicad
Identifier:            org.kicad.kicad
Version:               8.0.5 (8.0.5)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-09-17 02:59:30.5148 -0700
OS Version:            macOS 12.6.5 (21G531)
Report Version:        12
Anonymous UUID:        3E9CA79D-3423-4873-976A-2EA9582F85BB

Time Awake Since Boot: 890 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000030
Exception Codes:       0x0000000000000001, 0x0000000000000030
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [1099]

VM Region Info: 0x30 is not in any region.  Bytes before following region: 140737487368144
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      VM_ALLOCATE              7ffffff0f000-7ffffff10000 [    4K] r-x/r-x SM=ALI  

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libmupdf.dylib                	       0x120629430 fz_drop_icc_profile + 16
1   libmupdf.dylib                	       0x12062bf58 fz_drop_colorspace_imp + 136
2   libmupdfcpp.so                	       0x11e6dcdad mupdf::ll_fz_drop_colorspace(fz_colorspace*) + 61
3   libmupdfcpp.so                	       0x11e69d3bc mupdf::FzColorspace::~FzColorspace() + 12
4   _mupdf.so                     	       0x1221183a6 _wrap_delete_FzColorspace(_object*, _object*) + 70
5   _mupdf.so                     	       0x122359c0d SwigPyObject_dealloc(_object*) + 205
6   Python                        	       0x103bc610e dict_dealloc + 142
7   Python                        	       0x103be7d72 subtype_dealloc + 1106
8   Python                        	       0x103bc610e dict_dealloc + 142
9   Python                        	       0x103be7d72 subtype_dealloc + 1106
10  Python                        	       0x103bc37d1 PyDict_Clear + 465
11  Python                        	       0x103bc67e9 dict_tp_clear + 9
12  Python                        	       0x103ce2e8d collect + 3245
13  Python                        	       0x103ce21c8 _PyGC_CollectNoFail + 72
14  Python                        	       0x103ca5aeb _PyImport_Cleanup + 3483
15  Python                        	       0x103cbca40 Py_FinalizeEx + 160
16  kicad                         	       0x100614ad0 pybind11::finalize_interpreter() + 704
17  kicad                         	       0x1006146fd SCRIPTING::~SCRIPTING() + 29
18  kicad                         	       0x10047da3f PGM_BASE::~PGM_BASE() + 335
19  libsystem_c.dylib             	    0x7ff808d44de4 __cxa_finalize_ranges + 409
20  libsystem_c.dylib             	    0x7ff808d44bfe exit + 35
21  libdyld.dylib                 	    0x7ff808e58375 dyld4::LibSystemHelpers::exit(int) const + 11
22  dyld                          	       0x10246e558 start + 504

Thread 1:
0   libsystem_pthread.dylib       	    0x7ff808e47f48 start_wqthread + 0

[... 485 lines cut ...]
  ]
}
}

Model: VMware20,1, BootROM VMW201.00V.20904234.B64.2212051119, 8 processors, Unknown, 3.62 GHz, 8 GB, SMC 2.8f0
Graphics: Display, 128 MB
Display: spdisplays_display, 2545 x 1383, Main, MirrorOff, Online
Memory Module: RAM slot #0/RAM slot #0, 8 GB, DRAM, 4800 MHz, VMware Virtual RAM, VMW-8192MB
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Serial ATA Device: VMware Virtual SATA Hard Drive, 96.64 GB
Serial ATA Device: VMware Virtual SATA CDRW Drive
USB Device: USB20Bus
USB Device: USBBus
USB Device: USB31Bus
USB Device: VMware Virtual USB Hub
USB Device: VMware Virtual USB Hub
USB Device: VMware Virtual USB Keyboard
USB Device: VMware Virtual USB Mouse
Thunderbolt Bus: 
```
</details>

Last working version is 1.23.8, you can downgrade using this command:
`pip3 install --force-reinstall -v pymupdf==1.23.8`

denisbondar, 2024-09-17:
**It worked!**
Now all plugins are working fine.
Maybe we should pass this issue to the pymupdf project? Or somewhere that could solve it?

qu1ck, 2024-09-17:
I just filed one.