openscopeproject/InteractiveHtmlBom #468: Silkscreen not showing completely

doadoaf24, 2024-09-19:
Thank you for this tool, I started sing it and noticed that a few marks pointing to pin 1 in the silkscreen are not showing
here is the 3d view
[image]
this is the pcb
[image]
this is the tool you can see the small triangle pointing to pin 1 is not there
[image]

this is happening with a few other footprints in the board.

doadoaf24, 2024-09-19:
This is the footprint triangle
[image]

qu1ck, 2024-09-19:
Duplicate https://github.com/openscopeproject/InteractiveHtmlBom/issues/465

This is fixed already but not released yet.

closed 2024-09-19 by qu1ck