openscopeproject/InteractiveHtmlBom #471: Another nightly error

arturo182, 2024-10-12:
KiCad Version: 8.99.0-2584-g661ed81e6f, release build
Plugin: latest master (5915a24)

```python
Traceback (most recent call last):

  File "E:\Documents\KiCad\8.99\3rdparty\plugins\org_openscopeproject_InteractiveHtmlBom\ecad\kicad.py", line 864, in Run
    ibom.run_with_dialog(parser, config, logger)

  File "E:\Documents\KiCad\8.99\3rdparty\plugins\org_openscopeproject_InteractiveHtmlBom\core\ibom.py", line 363, in run_with_dialog
    main(parser, config, logger)

  File "E:\Documents\KiCad\8.99\3rdparty\plugins\org_openscopeproject_InteractiveHtmlBom\core\ibom.py", line 327, in main
    pcbdata, components = parser.parse()
                          ^^^^^^^^^^^^^^

  File "E:\Documents\KiCad\8.99\3rdparty\plugins\org_openscopeproject_InteractiveHtmlBom\ecad\kicad.py", line 781, in parse
    "footprints": self.parse_footprints(),
                  ^^^^^^^^^^^^^^^^^^^^^^^

  File "E:\Documents\KiCad\8.99\3rdparty\plugins\org_openscopeproject_InteractiveHtmlBom\ecad\kicad.py", line 564, in parse_footprints
    pad_dict = self.parse_pad(p)
               ^^^^^^^^^^^^^^^^^

  File "E:\Documents\KiCad\8.99\3rdparty\plugins\org_openscopeproject_InteractiveHtmlBom\ecad\kicad.py", line 490, in parse_pad
    pad_dict["radius"] = pad.GetRoundRectCornerRadius() * 1e-6
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

TypeError: PAD.GetRoundRectCornerRadius() missing 1 required positional argument: 'aLayer'
```

Fixed locally by changing:
```diff
--- a/ecad/kicad.py
+++ b/ecad/kicad.py
@@ -487,7 +487,7 @@ class PcbnewParser(EcadParser):
             ]]

         if shape in ["roundrect", "chamfrect"]:
-            pad_dict["radius"] = pad.GetRoundRectCornerRadius() * 1e-6
+            pad_dict["radius"] = pad.GetRoundRectCornerRadius(layers_set[0]) * 1e-6
         if shape == "chamfrect":
             pad_dict["chamfpos"] = pad.GetChamferPositions()
             pad_dict["chamfratio"] = pad.GetChamferRectRatio()
```

But not sure if that's the right way to do it.

qu1ck, 2024-10-12:
It's a decent workaround for now but proper fix needs padstack support. But kicad's own implementation is not yet complete so I'll wait on this.

Bowlsnapper, 2024-10-12:
Yeah, I have the same issue. Definitely a bummer, but I hope a fix can be found! :)

qu1ck, 2024-10-12:
You need to reload kicad after you change the plugin (or do plugin refresh from the pcbnew menu). If you still have an issue it should not be exact same one, show me your error message.

Bowlsnapper, 2024-10-13:
[image]

qu1ck, 2024-10-13:
I suspect your nightly is old, I fixed that on kicad side a while ago.

Bowlsnapper, 2024-10-13:
Hmmm, lemmie update and see.

Bowlsnapper, 2024-10-13:
Yes, you are correct. Installing the latest nightly fixed the issue. You are a god. I love your work.

robomechs, 2024-10-14:
for me it didn't  help.
Application: KiCad x64 on x64
Version: 8.99.0-2659-gf7f86174a0, release build
[image]

robomechs, 2024-10-14:
> Yes, you are correct. Installing the latest nightly fixed the issue. You are a god. I love your work.

Which version have you used for success?

qu1ck, 2024-10-14:
You either did not apply the modification correctly or did not reload the plugin or restart kicad. Your screenshot shows current code, not modified one.

robomechs, 2024-10-14:
Seems like I didn't apply modification correctly, becayse I installed last KiCad version and reload the computer just in case.

robomechs, 2024-10-14:
from box I've got this error:
[image]
arter that I download last ibom from github and put it instead of the one from "box", reload KiCad, and
[image]

qu1ck, 2024-10-14:
Yeah and to fix that last error you need to change the code like the first post shows.
Change this line in `kicad.py`
`pad_dict["radius"] = pad.GetRoundRectCornerRadius() * 1e-6`
to
`pad_dict["radius"] = pad.GetRoundRectCornerRadius(layers_set[0]) * 1e-6`

commit 70e2a5e3 (2025-04-08): Add padstack support for kicad 9

Fixes #471
closed 2025-04-08 by qu1ck