openscopeproject/InteractiveHtmlBom #473: Support for manufactures part numbers.

gnfontaine, 2024-10-18:
The plugin is great, but when ordering from manufactures you need to include the manufactures part numbers. Can this be added? Thanks

qu1ck, 2024-10-18:
You can add anything you want as custom field. Add your part numbers in eeschema, update the pcb and enable the field in ibom config.

domleblanc94, 2024-10-31:
> You can add anything you want as custom field. Add your part numbers in eeschema, update the pcb and enable the field in ibom config.

I seem to be having an issue with this when trying to run from the CLI, when using either the --show-fields or --extra-fields commands they appear in the BOM headers but all cells are blank. If I run fields using the Plugin it populates everything. Have I missed something? Also the --dnp-field "kicad_dnp command does not appear to work though the CLI.

qu1ck, 2024-10-31:
What does your full command look like? Don't forget to specify the `--extra-data-file` flag (usually same as pcb).

mMuck, 2024-11-25:
Could this be closed? There is no real issue, right? 😉

closed 2024-11-25 by qu1ck

gnfontaine, 2024-11-25:
Yes

-- 
Gerry Fontaine
180 Hall Rd, Headingley, MB R4H 1A3
Cell  : 204-588-8835

The information in this electronic transmission (including any attachments)
may contain confidential or legally privileged information and is intended
solely for the individual(s) or entity(ies) named above. If you are not an
intended recipient, you are hereby notified that reading, distributing, or
otherwise disseminating or copying, or taking any action based on the
contents of this transmission is strictly prohibited. Any unauthorized
interception of this transmission is illegal under the law. If you have
received this transmission in error, please immediately notify the sender
by return email and then destroy all copies of the transmission.