openscopeproject/InteractiveHtmlBom #475: Unused elements' silkscreen isn't removed

PrzemekBarczyk, 2024-10-28:
I'm using your project to generate interactive BOMs for the EAGLE program. I've noticed discrepancies in the output file depending on whether the input file is a .brd file or a .json file (generated with brd2json.ulp).

If a project has a variant with hidden components, the BOM generated from the .json file does not include these components in either the table or the schematic. In contrast, for .brd files, the hidden components are only excluded from the table but remain visible on the schematic.

Below, I’ve attached screenshots to illustrate these differences.

Are the described differences intentional design decisions? Does the project allow for variant configuration using .brd files so that hidden components are also excluded from the schematic?

Thank you!

[image]
BOM obtained using .json

[image]
BOM obtained using .brd

qu1ck, 2024-10-28:
Silkscreen is expected to not be removed because it does not magically disappear from the board if you don't solder the component. The rendering is supposed to represent how the actual board looks, including pads and silk markings.

Not sure why using the json method shows that behavior but that method ulp is not maintained afaik, use native .brd support.

PrzemekBarczyk, 2024-10-28:
Thanks for the quick response!

closed 2024-10-28 by PrzemekBarczyk

Funkenjaeger, 2024-11-01:
I've added an issue to the brd2json repo regarding the incorrect behavior with the ULP (footprints missing for unpopulated components) - just for posterity, as I'm not inclined to fix it presently.  As @qu1ck  said, the ULP is essentially OBE at this point due to native .brd/.fbrd support in ibom which has been in place for just shy of 2 years now, so I've gone ahead and archived that repo.