openscopeproject/InteractiveHtmlBom #482: Some silkscreen & fab layer elements rendered at wrong place outside of PCB outline

mMuck, 2024-11-26:
Naturally I confirmed the positions of the elements in the PCB project but there they are at correct places.
In the iHtmlBom Canvas they're rendered even somehow rotated. Are there known conditions for such cases?
 
[image]

If this should be investigated I can try to strip down the project to only this elements. Or a snippet of the relevant kicad_pcb file parts if it disappears with less parts.
Compression & zones on, nets off. Current KiCad 8.0.6 (Win10) & latest PCM release of iHtmlBom.

qu1ck, 2024-11-26:
Already fixed in 002d62760ae2c91a267d8c8d2a3b5eccf81439a2 but it's not released yet.

mMuck, 2024-11-27:
Wow. Thanks. I see - there are reasons to test the latest master. I'll do that and give feedback.

Lambosaurus, 2025-02-23:
Hey, I see this is still an issue. Could you do a release to get this fix live?

qu1ck, 2025-04-10:
Version 2.10.0 is now released.

closed 2025-04-10 by qu1ck