openscopeproject/InteractiveHtmlBom #487: Altium Designer Support

zharovdv, 2025-01-20:
I made InteractiveHtmlBom based Altium Designer [plugin](https://github.com/zharovdv/InteractiveHTMLBOM4Altium2). Due altium script system limitation/performance now it works standalone. It's generate little bit easy json format, assemble it with js preprocessor and InteractiveHtmlBom files. Is any possibility be mentioned on Compatibility section of your wiki or it’s should generate json file?

PS: Yes, i know about boastools.com - but i cant use it due altium version incompatibilities.

zharovdv, 2025-01-20:
Looks like I found out solution in a way EasyEDA works. Sorry for spend your time. I'll be back with pull request soon.

qu1ck, 2025-01-20:
Keep in mind that I will not be able to test or maintain a custom parser for Altium because I don't have it. 

Instead look at how Allegro is implemented, there is a separately maintained project that exports allegro pcb data into an intermediary json format that InteractiveHtmlBom can work with. This is the recommended way to integrate non-free software with InteractiveHtmlBom.

zharovdv, 2025-01-21:
Sure, I understand. Looks done. https://github.com/openscopeproject/InteractiveHtmlBom/pull/488

qu1ck, 2025-01-21:
Sorry, I can not merge that as I will not be able to maintain it, as explained above.

Change your pascal plugin code to format json according to ibom schema and you will get ibom compatibility for free.

Funkenjaeger, 2025-01-22:
At a glance, it looks like @zharovdv may need to extend the generic JSON schema and parser to support the "rectangle pad hole" option that they added as part of #488, if that's something actually needed for Altium compatibility.

zharovdv, 2025-01-22:
> At a glance, it looks like [@zharovdv](https://github.com/zharovdv) may need to extend the generic JSON schema and parser to support the "rectangle pad hole" option that they added as part of [...]

@qu1ck already add "rectangle pad hole", it was in last pull request. I don't know why github show this commit again... So, in any way I will try to change my altium script generate "schema" json.

Funkenjaeger, 2025-01-22:
Ah, I see that now, I missed that recent change - disregard my comment!

zharovdv, 2025-01-22:
> Sorry, I can not merge that as I will not be able to maintain it, as explained above.
> 
> Change your pascal plugin code to format json according to ibom schema and you will get ibom compatibility for free.

Done. Now my script generate "genericjsonpcbdata_v1.schema" compatible file.

qu1ck, 2025-01-22:
Great!
Can you add usage instructions on readme of your plugin? Also an example output would be helpful, if AD has some kind of well known demo board it would be best to use that as source.
I'll add a link to it in wiki.

zharovdv, 2025-01-23:
Sure, will be done in a few days.

zharovdv, 2025-01-23:
BTW, is DrawingText in schema correct? I dont see many properties (text, pos, angle...) in it. Looks like it's out of date.

qu1ck, 2025-01-23:
Yes, it's up to date, svgpath text rendering method is the newest and preferred way. It does not need any of the other fields.

If you can't generate an svgpath for some reason then you can add `polygons` and `pos` to the schema but don't use the other stuff, it's only there for compatibility with old kicad versions.

zharovdv, 2025-02-02:
> Great! Can you add usage instructions on readme of your plugin? Also an example output would be helpful, if AD has some kind of well known demo board it would be best to use that as source. I'll add a link to it in wiki.

Finally done. Hope it's clear. GIF instruction added too. I can't add AD demo project cause of copyrights. Now looking for opensource AD project, enought small, but compilcated and with appropriate license.

qu1ck, 2025-02-03:
Looks good, thanks. I updated the wiki.

closed 2025-02-03 by zharovdv