openscopeproject/InteractiveHtmlBom #491: Error in grouping components

enboer, 2025-02-03:
I have an issue with grouping components. On the command line I specify --group-fields to by "PART_NUMBER". But in the tabel it groups on 'all' the fields that are shown. So in my case also the Y column needs to be the same. I want them to be grouped on the PART_ NUMBER. How do I need to do that?

[image]

qu1ck, 2025-02-03:
What is your full command line?
Can you show me the output file? Strip the board to the bare minimum that shows the issue if you need to keep the design confidential.

enboer, 2025-02-04:
this is the command.
C:\appl\cadence\custom\pcb\utils\InteractiveHtmlBom\generate_interactive_bom.py "D:/Work/Demos/Demo_174/allegro\json\demon-complete-xx_LowEnd.json" --name ..\..\output\demon-complete-xx-LowEnd-90 --group-fields "PART_NUMBER" --extra-fields "Y"

I attached the .json file (from Allegro) and the output HTML file.

[demon-complete-xx_lowend.json](https://github.com/user-attachments/files/18653858/demon-complete-xx_lowend.json)

[demon-complete-xx-LowEnd-90.zip](https://github.com/user-attachments/files/18653862/demon-complete-xx-LowEnd-90.zip)

qu1ck, 2025-02-04:
Don't use `--extra-fields` if you use `--group-fields`, it effectively adds `Y` to the group fields because it is equivalent to setting both `--group-fields` and `--show-fields`.
Instead use `--group-fields PART_NUMBER --show-fields Value,Footprint,PART_NUMBER,Y`.

enboer, 2025-02-04:
Thanks that works. Initially I used extra-fields only, but then grouping is not working either. From the help it looks like using extra-fields is the same as show-fields and group-fields together.  But that is not the case?

enboer, 2025-02-04:
One other question. if you hide columns,  or parts (via the checkboxes) this is stored in the browser. What needs to be reset to be able to clear this. It is not only 'cached data'

qu1ck, 2025-02-04:
Settings and checkboxes are saved in browser local storage. You can reset everything from menu

[image]

closed 2025-02-09 by qu1ck