openscopeproject/InteractiveHtmlBom #493: Jobset usage iBom KiCad 9

jrzwee, 2025-02-18:
When the iBom is used with the new jobset feature of KiCad 9, the "extra-fields" is not working correctly.

[image]

With the job:

[image]

python "PATH-TO-PYTHON\generate_interactive_bom.py" "PATH-TO-PCB\UI.kicad_pcb" --extra-fields "MF,MPN" --name-format "${PNAME}_${GREV}_iBOM_${CURRENT_DATE}" --dest-dir "production"

[image]

When performed with the KiCad plugin from the PCB editor:

[image]

jrzwee, 2025-02-18:
Other flags I tested which are not working:

- checkboxes 
- include-tracks
- extra-fields

jrzwee, 2025-02-18:
MPN and MF are fixed when I add --extra-data-file ${PROJECTNAME}.kicad_pcb

The other problems are not.

qu1ck, 2025-02-18:
Checkboxes and include tracks will be cached in browser, to see those changed you need to reset settings in html page (it's in export menu).

> show-dialog (dialog shows "running job" and takes forever)

Are you sure the ibom config dialog is not hidden behind other windows? It will not be part of kicad window group or capture focus if it is run as part of jobset, it will be a separate application.

jrzwee, 2025-02-18:
Checkboxes and include tracks work exactly how you described it. Thank you for the help on those.

> Are you sure the ibom config dialog is not hidden behind other windows? It will not be part of kicad window group or capture focus if it is run as part of jobset, it will be a separate application.

It seems the application freezes; I am not able to click anywhere or move the window.

[image]

This is what I can see.

qu1ck, 2025-02-18:
It is expected that kicad will not do anything, it is waiting for ibom to complete and ibom is waiting for your user input on the config dialog. Try alt-tab to find the ibom config dialog and confirm (or cancel) bom generation, then kicad will continue with the job.

qu1ck, 2025-02-18:
That said, kicad should not freeze completely, the timer should count and cancel button should work, if that is not the case it's a kicad bug.

jrzwee, 2025-02-19:
Alt-Tab does not show any popups unfortunately.

This is what I can see:

[image]

I guess i need to create a KiCad issue. 👍

jrzwee, 2025-02-19:
https://gitlab.com/kicad/code/kicad/-/issues/19997

qu1ck, 2025-02-19:
Thanks, I did some tests and commented there with my findings. This is not an ibom issue.
Hopefully kicad will fix this soon, in the meantime avoid using the ibom gui in jobset, you can configure everything directly through ibom's command flags.

closed 2025-02-19 by qu1ck