openscopeproject/InteractiveHtmlBom #499: Silkscreen logo shown on wrong side of the board

gsuareznasa, 2025-03-14:
I have a few logos that i insert as footprints not in the schematic and it seems that the ibom output shows them in the wrong side of the board. I'm using InteractiveHtmlBom v2.9. Any idea what I'm doing wrong here?

TOP
[image]
[image]

BOTTOM

[image]
[image]

Thanks.

qu1ck, 2025-03-14:
Probably already fixed in 002d62760ae2c91a267d8c8d2a3b5eccf81439a2 can you try installing latest from git? 
Uninstall package from plugin manager, download https://github.com/openscopeproject/InteractiveHtmlBom/archive/refs/heads/master.zip
extract into folder you can open from `Tools → External Plugins → Open Plugin Directory` in pcb editor, restart kicad (or refresh plugins from the same menu).

gsuareznasa, 2025-03-14:
I just tried it and it works now. I passed this info to my coworkers using this plugin. Thanks!

qu1ck, 2025-03-14:
Thanks for confirming, don't forget to remove the manual installation when new version with the fix is released and you switch to plugin manager package again.

I'll ping this issue when the fix is published.

Kutilak, 2025-03-25:
> Thanks for confirming, don't forget to remove the manual installation when new version with the fix is released and you switch to plugin manager package again.
> 
> I'll ping this issue when the fix is published.

Thank you, I will wait for the official fix release

qu1ck, 2025-04-10:
Version 2.10.0 is now released with the fixes.

closed 2025-04-10 by qu1ck