openscopeproject/InteractiveHtmlBom #500: Access Kicad's text variables values from --name-format parameter

tbornon-sts, 2025-03-18:
Hello,

I was wondering if it is possible (but then I didn't find the information on the wiki) to use one of Kicad's text variables to set the name of the ibom from the CLI. If it isn't, do you think it would be doable to add the feature ?

Regards,

qu1ck, 2025-03-18:
You can't use text variables directly but you can use metadata block, there is help message in gui

[image]

And you can use text vars inside metadata fields.

tbornon-sts, 2025-03-19:
I've seen these parameters however, I have my part number which is stored in text vars. I'm not willing to put it in any of the available metadata fields of the PCB.
I will try to find another solution if it isn't possible then.

Thanks for your help

qu1ck, 2025-03-19:
I looked into adding special substitution for kicad vars and while possible it will be not portable to other EDA tools. Since there are easy workarounds I will not add that.

Text vars are stored in kicad project file which is plain json. You can easily extract value of text var using `jq` or any other similar tool
```bash
$ jq .text_variables.FOO blinky_tree.kicad_pro
"BAR"
```
Chain that into sed to strip quotes and you can put it directly into --name-format flag.

closed 2025-03-22 by qu1ck