openscopeproject/InteractiveHtmlBom #508: v2.10 renders NPTHs in SMD footprints differently

mMuck, 2025-05-19:
Thanks for the new release!
I found out that there are a differences a SMD footprint containing NPTHs. Any ideas what has changed?

v2.9 NPTH visible
[image]

v2.10 NPTH missing
[image]

qu1ck, 2025-05-19:
I don't see this issue in my test files for kicad 9.0.2. Please share your board (you can remove everything else, just this one footprint is enough) and your ull kicad version.

mMuck, 2025-05-20:
[debug-project.zip](https://github.com/user-attachments/files/20349929/debug-project.zip) (v9.0.2)
I copied the FP to a new empty project, exported the ibom - problem persists. Other NPTHs in the test board are rendered as normal!?

qu1ck, 2025-05-21:
Thanks, should be fixed now.

Those pads are different than usual because they are marked as having no copper (new feature in kicad 9 related to padstacks). My code would ignore such pads but it does not make sense to ignore pads with holes regardless if they have copper on relevant layers.

closed 2025-05-21 by qu1ck

mMuck, 2025-05-21:
👍 Thanks a lot. That was quick 🙂

commit f42ee38f (2025-05-21): Dont ignore through hole pads even if they have no copper on F/B layers

Fixes #508