openscopeproject/InteractiveHtmlBom #518: Using webserial API to control manual P&P systems - just an idea.

crazy4volvo, 2025-09-24:
For me it was quite a discovery despite many years dealing with electronics... 

   Between pure manual prototype assembling and fully automatized production pick-and-place systems  there is a room for machines designed for supporting manual prototype assembly. Probably the most advanced are the devices from https://www.fritsch-smt.de . 

   I found some attempts to guide iBom users  with projecting the image onto a PCB and highligtning the actual assembly position, but I came with different idea:
 
 Could it be possible to transmit the xy coordinates of currently selected part via webserial? This would open a lot of possibilities for hardware freaks (like me ;) ) to build simple motorized devices guiding through PCB assembly process with a spotlight, laser dot, two crossed line lasers etc. or even more advanced ones with mechanically guided needles, moving cameras etc.  If the interface would allow to emulate Up and Down commands coming from footswitches connected to the controller, the technician's hands would be free from constantly reaching to mouse/keyboard during the assembly process.

As a person preparing to assemble two boards containing ca. 1200 parts (half of them are 0603), I would really appreciate such a feature ;).

Funkenjaeger, 2025-09-24:
In my opinion, this seems like way too niche of a use case to be integrated into ibom.  
A more reasonable starting point could be to explore whether adding some sort of general-purpose web API to ibom might make sense, and then you could implement your own external application-specific tool that integrates via that API and performs the functions you outlined.

qu1ck, 2025-09-28:
Yes, this is very niche and also why custom javascript with a simple event system was added.
You can add a listener for the highlight event and receive a list of part ids/refs which are highlighted, look up their coordinates in pcbdata and do whatever you want with that, including sending to a socket, serial, etc.

https://github.com/openscopeproject/InteractiveHtmlBom/wiki/Customization#custom-javascript

closed 2025-09-28 by qu1ck

crazy4volvo, 2025-09-28:
Thanks for the answers. I'm afraid I'll give up , this task is way above my level.